OpenVSP/OpenVSP #228: Wrong results without enabling X-Z symmetry

Vinc83, 2022-05-05:
Hi,
 it seems that if you don't enable X-Z symmetry, the tool provides wrong results.

ramcdona, 2022-05-05:
This comment is entirely vague and utterly useless.  Please try to describe what you're actually doing, what you expect to happen vs. what does happen.

If you're a new user, the Google Group is probably the best place to ask for and receive help.

closed 2022-05-05 by ramcdona

Vinc83, 2022-05-06:
I am doing an aerodynamic analysis on a symmetric configuration using OpenVSP 3.27.1. VSPAERO tool.
If I enable, in the "advanced tab", the "X-Z symmetry" checkbox I have a result.
For example, in my case, I have a CL=0.35 (which is coherent with the results I have obtained with another tool).
If I don't enable the "X-Z symmetry" checkbox I have a different result, in my case CL = 0.56.
I hope that now the issue is more clear.