OpenVSP/OpenVSP #234: spar is not modelled correctly in OpenVSP 3.28.

FanglinYu, 2022-07-12:
Hello,

The input file is exactly same. As you can see from the first picture, the spar is not modelled correctly in version 3.28, compared with version 3.27(second picture).

[image]

[image]

ramcdona, 2022-07-25:
Can you please post or email me the *.vsp3 file corresponding to this test case?

FanglinYu, 2022-08-01:
Hi,

Thank you for the reply. Attached is the vsp3 file.
Regards,
Fanglin

ramcdona, 2025-07-19:
Please try in a recent version (3.43.0 or newer).  You might want to check out the new Poly Spar structural component, it is better than the old Spar in many ways.