OpenVSP/OpenVSP #243: pip install leads to "VSPAero solver not found"

jeffallen, 2022-11-25:
After following the instructions in /opt/OpenVSP/python to do an install into a Conda environment, we found that the vspaero, vspviewer and vsploads binaries are not present in the $HOME/.conda/envs/vsppytools/lib/python3.10/site-packages/openvsp directory.

The binaries were made by CMake correctly, and ended up in the RPM, and the RPM put them in /opt/OpenVSP/python/openvsp/openvsp. The place they get lost is when `pip install -r requirements.txt` uses `setup.py` to figure out what to install. So it seems like the right magic to tell `setup.py` about non-Python elements of the install is missing.

cc @mumbrict

ramcdona, 2022-12-18:
Please try adding these three lines to the MANIFEST.in file in python/openvsp/openvsp before the package installation.

include openvsp/vspaero*
include openvsp/vsploads*
include openvsp/vspviewer*

Theoretically, that should cause them to be copied along with all the *.py files.

The * is there to hopefully work on both Linux/MacOS (where it should match with nothing) and on Windows (where it should match with .exe).

ramcdona, 2023-03-02:
This should be addressed in 3.32.0.

closed 2023-03-02 by ramcdona