OpenVSP/OpenVSP #244: Issue with trimmed igs export to in-house CAD package

jrj1141367, 2022-12-02:
Hi,

Found a potential issue with the trimmed .iges export utility for surfaces. Issue has become apparent when bringing them into our in-house software which demonstrates some odd behaviour with the imported surfaces.

One of our SME's has tracked the issue down to the following:

"I’ve tracked the problem down to the BSpline (NURBS) surface knot vector definition. The knot vectors are lists of non-decreasing real numbers in each parametric direction. For a nth order spline the first and last set of knots must be repeated n+1 times.

So for example a 3rd order NURBS knot vector should be something like:

0.0 , 0.0 , 0.0 , 0.0 , 1.0 , 2.0 , 3.0 , 4.0 , 4.0 , 4.0 , 4.0.

OpenVSP is writing the surface knot vectors with the final knot value incremented by 1.0; e.g.

0.0 , 0.0 , 0.0 , 0.0 , 1.0 , 2.0 , 3.0 , 4.0 , 4.0 , 4.0 , **_5.0_**.

The highlighted value is incorrect by any standard & is effectively corrupting all of the surfaces (the error doesn’t seem to be there in the curves)."

ramcdona, 2022-12-17:
Thanks for bringing this to my attention.  Sorry I haven't noticed it until now.  I generally track the Google group more closely (good for quick response) than the issue list -- but this is a good place to make sure things don't get lost over time.

Does the same behavior show for non-trimmed IGES files?  I.e. those exported from the File... Export... menu?

ramcdona, 2023-01-21:
I can not duplicate this problem locally.  Can you provide a simple *.vsp3 file that exhibits this problem?  Please email the file off-line.  I can sign an NDA if needed.

jrj1141367, 2023-02-06:
Hi, apologies for the delayed response. Yes it is occurring with untrimmed iges files as well. Happy to email on an example of the behaviour, can you supply an email address I can send on to? No NDA required as we can use a general case to replicate the issue.

ramcdona, 2023-02-06:
Please send the file to rob dot a dot mcdonald at gmail dot com.  Please include the *.vsp3 file, exact steps to recreate the bug, an example of the broken *.iges file, and the description of what you think should be different about the file.

I tried some trivial example files and could not see this problem or where it might be coming from.  Hopefully your test will clear up what is happening.

ramcdona, 2023-02-14:
I am trying to address a bunch of smaller issues like this one.  Hopefully I can work solutions by the end of the week for a next release.  If you can provide a sample case, I may be able to get this fixed.  Otherwise, I have been unable to duplicate the problem locally.

commit e2ea0fa7 (2023-10-28): Fix issue with IGES knot values.

Fixes #244
closed 2023-10-28 by ramcdona