OpenVSP/OpenVSP #245: ccx and cgx bit found

zufus, 2022-12-03:
Hi,

when I launch vsp, a terminal window appears, and gives the following messages (I am translating them from Italian):

"ccx" not recognized as command, neither internal or external, and executable program or batch file
"cgx" not recognized as command, neither internal or external, and executable program or batch file"

This happens on openVSP version 3.31.1, Windows 10 64 bit.

jeffallen, 2023-01-31:
This is a normal message, and nothing to worry about.

But maybe it would be better to remove these messages on startup and only show them later when the FEM analysis code actually tries to use the programs?

ramcdona, 2023-04-29:
3.33.0 includes a fix that should silence this message.

closed 2023-04-29 by ramcdona