OpenVSP/OpenVSP #247: NASTRAN output violates the spec

jeffallen, 2023-01-23:
The current NASTRAN output format is [comma-separated](https://github.com/OpenVSP/OpenVSP/blob/1e1d63c1ec19cd9e5c8eb48eb5cbcc89c878101c/src/cfd_mesh/FeaElement.cpp#L350).

As a result I get CQUAD4's like this:
```
CQUAD4  ,    2279,       1,    1321,    2161,    2165,    1322,161.9520
```

According to the [Siemens NASTRAN manual](https://docs.plm.automation.siemens.com/data_services/resources/nxnastran/11/help/tdoc/en_US/pdf/QRG.pdf), section 9.3, there are certain rules for "free field format":
* use commas to separate columns
* integers and character input must be 8 characters or less
* no embedded blanks are allowed
* (there are other rules about continuation and comments, but those are not interesting to us)

I plan to submit a PR that corrects the NASTRAN output, once I have managed to get my files to load into FEMAP.

jeffallen, 2023-01-24:
Today I managed to get the syntax errors in the NASTRAN file cleared up, but FEMAP is still not loading it, so I won't post a PR until I can actually get FEMAP to accept it.

jeffallen, 2023-01-31:
To be clear: It is possible there are more changes there than strictly required. But I certainly had problems with FEMAP giving errors with the -0.00000 formulation of 0, and with the element id of 0 (the error message it gives it not clear, but when I finally tried starting the element id's at 1, that problem went away, letting me discover the second one).

We are still having some trouble with the generated mesh, where certain nodes displace themselves in unreal ways when we put a load on the wing tip. If we can create a useful repro case, we will open a different issue with that.

ramcdona, 2023-03-02:
This is hopefully resolved in 3.32.0.  Please report back if you continue to see problems like this.

closed 2023-03-02 by ramcdona