OpenVSP/OpenVSP #250: With FEA units "MPA" the units of NASTRAN export are mixed

jeffallen, 2023-02-02:
When FEA Structure -> Structure -> Unit system is set to MPA, then a mesh is generated, then the mesh is exported, the general magnitude of the GRID lines correspond to the unitless measurements of (for example) the wing length. However, the property (DefaultShell, for example) has the thickness set in mm. So for a wing that is "9" long, the max GRID x you'll see is 9.000, while the thickness you will find in the PSHELL lines is for example "3.000", for 3 mm.

When a file like this is imported into FEMAP, the thickness and the size of the model are off by a factor of 10^3, because the wing is 9 meters long, and the skins are assumed by FEMAP to be 3 meters thick.

Our FEMAP expert says that she really wants all the GRID points to be specified as MM, so that the size of the model corresponds to the rest of the material parameters, which are also in MMs in the MPA system.

I am investigating this issue more to understand if we are using OpenVSP wrong, or if there is a real problem here.

cc @mumbricht

jeffallen, 2023-02-06:
Marc helped me understand that in the rest of OpenVSP (i.e. other than FEA), the unit of length is "undefined but consistent". For example, in his work, he makes the chord of the wing 0.63, because he plans for the real wing to be 63 cm wide. Then all the other parameters are consistent, all as m (or m^2, etc).

When we export the FEM mesh and choose "MPA" as the units for our materials and the thicknesses in our properties, we are telling OpenVSP's NASTRAN output code, "you know all those grid points you are getting from the model? They are in mm's." And that's not right.

So I asked Marc, "well, what if you changed to modeling in mm?" And he doesn't like that, because so many supporting calculations and other stuff is in meters. In his mind, modeling happens in the unit that's close to the size of the aircraft, and FEA happens in mm for FEA's own reasons.

That means we are considering to add a UI element in the FEA Mesh (maybe next to Structure->Unit System) that's a checkbox and a popup. The checkbox would indicate "The model is using a different length unit than the chosen FEA Unit system", and the popup would indicate what the model's unit are. The default for the checkbox would be false, and the popup would be disabled when the checkbox is not checked.

In our case, we'd choose "Unit system: MPA, Model units need converting: true, Model units: Meter".

ramcdona, 2023-02-06:
You've correctly assessed the situation.  OpenVSP (for better or worse) is generally units agnostic.  When we come to an analysis that cares, we let the user specify what they were thinking at the time.

I generally like your suggestion.  I would probably label the pull-down menus "Model Units" and "FEM Units".  I don't know that a separate checkbox is needed -- but having one will make legacy files behave as-is in the first go.

The next question is _when_ to convert the model.  There are two 'easy' answers.  1) at the very start of the meshing process, when the surfaces are transferred from OpenVSP to the FEM stuff.  2) at the very end, just as the files are written out.

The meshing parameters (things like max/min edge length) are in the model dimensions -- so even though Marc's model is in m, I suspect your FEMAP expert would rather specify mesh parameters in mm.

Sometimes OpenVSP's intersection and meshing tools have tolerances that end up relative to the model scale (we try to nondimensionalize them wherever possible).  And sometimes that leads to models that work at one scale, but fail at another.  I need to fix those bugs (so if you find a repeatable one, please share the files), but I can't guarantee that you won't see some frustration in this process.

I will be working in this area between now and May - I could add this to my list and it would be pretty simple for me to take care of.  However, I can't promise any particular timeline in there right now.  I'm happy to help guide you in this as much as possible.

The immediate way to get your co-worker productive is for them to open Marc's file, select the components involved in the structure model and scale them in the OpenVSP context.  Then they can do their FEA work with the entire model in mm.  While this is another manual step, it only takes a moment.

ramcdona, 2023-03-02:
This has been addressed in 3.32.0.

closed 2023-03-02 by ramcdona