OpenVSP/OpenVSP #256: Parasite Drag Tool inop in v3.32

CaptNPlus9, 2023-03-03:
I have run several models in v3.32 Parasite Drag tool that report zero wetted surface area for fuselages and wings (attachment 1). Those same models give reasonable non-zero results in v3.30 (attachment 2).
[image]
[image]

ramcdona, 2023-03-05:
Thanks for bringing this to my attention, it will be fixed in the next release.

This issue tracker is a great place to make sure bug reports don't get lost over time -- but the Google Group is the best place to make sure they get noticed quickly.  I happened to check it today, but I don't get automatic notifications of new issues and sometimes go weeks between checking for new ones manually.

CaptNPlus9, 2023-03-05:
Rob,
Thanks for all your efforts in supporting and improving this capability.

I teach flight dynamics at the Naval Academy. My students each build a performance model in OpenVSP of their assigned Warbird in the spring semester of their junior year for our Performance course. Then they return in the Fall and refine their model for S&C to build a model they can upload to our sim. I have a TA to help them through the first iteration in the spring. They’re on their own in the fall. They’re then prepped to build models for their Capstone Design course throughout their senior year. The system works well within our curriculum.

I’ll remember that about Google Group. We’re in the Google ecosystem for email and apps, yet our institutional settings block Google Groups, for some reason.
Reaching the Google Group entails a work-around from home using a personal, vice work, account. 

V/R, Rob 

Robert J Niewoehner
CAPT, USN (ret), PhD
Professor, Aerospace Engineering
United States Naval Academy
***@***.***
410-703-5848(m)

commit 935f9c71 (2023-03-22): RefreshTagMaps instead of SubTagTris in CompGeomAndFlatten

Restore SubTagTris in IntersectTrim() was needed for ParasiteDrag

Fixes #256

Also fixes problems with inaccurate tagging.

Should still keep empty tags out of key files.
closed 2023-03-22 by ramcdona