OpenVSP/OpenVSP #273: Build error: missing include of time.h

jeffallen, 2023-08-21:
I found when building commit c011578175da93a5f03cb4 on Fedora 36, that I got this error:

```
/work/OpenVSP/buildlibs/DELABELLA-prefix/src/DELABELLA/delabella.cpp:56:23: error: ‘CLOCK_MONOTONIC’ was not declared in this scope
   56 |         clock_gettime(CLOCK_MONOTONIC, &ts);
      |                       ^~~~~~~~~~~~~~~
/work/OpenVSP/buildlibs/DELABELLA-prefix/src/DELABELLA/delabella.cpp:56:9: error: ‘clock_gettime’ was not declared in this scope
   56 |         clock_gettime(CLOCK_MONOTONIC, &ts);
      |         ^~~~~~~~~~~~~
```

I fixed it with:

```
$ diff -u DELABELLA-prefix/src/DELABELLA/delabella.cpp- DELABELLA-prefix/src/DELABELLA/delabella.cpp
--- DELABELLA-prefix/src/DELABELLA/delabella.cpp-	2023-08-21 12:37:27.968198000 +0200
+++ DELABELLA-prefix/src/DELABELLA/delabella.cpp	2023-08-21 12:37:46.823645000 +0200
@@ -24,6 +24,7 @@
 #include <algorithm>
 #include <stdint.h>
 #include <string.h>
+#include <time.h>
 
 #include "delabella.h"
 #include "predicates.h"
```

I did not investigate more deeply why this was needed, I was on my way to go investigate another problem at the time.

acxz, 2023-09-10:
I can also confirm this issue on Arch Linux

acxz, 2023-09-10:
Here is the upstream issue: https://github.com/msokalski/delabella/issues/11

closed 2023-10-28 by ramcdona