OpenVSP/OpenVSP #274: Failed STEP export of stack + rectangle edit curve

jeffallen, 2023-09-11:
The attached file shows a model that when exported as STEP, is corrupt. Other cases of "stack with xsections of point, edit curve (8 points in a rectangle), point" work, I have not been able to figure out what the difference is with this one. Something about the fact that the caps are flat, and that there is a sharp turn seems to trigger it.

jeffallen, 2023-09-12:
I investigated the divide by zeros, and they seem to be related to some AdvLink entries that were left in the .vsp3 accidentally. I am still investigating them, because they also caused a crash while debugging, but I don't think they are important to the failed STEP export.

jeffallen, 2023-09-12:
This vsp3 file shows the problem. When the strengths are not 0.0, the edge between the face and the sides is curved, and it exports correctly. When the strengths are 0, the edge is discrete (90 degrees) and the STEP export goes wrong.

[issue274a.zip](https://github.com/OpenVSP/OpenVSP/files/12586046/issue274a.zip)

jeffallen, 2023-09-25:
Sorry, I don't understand. The only recent commit on the Fixes branch is b9c425fa5a8c8aff8d5830bad9ea9f171d56fdc5, which is already applied to the main branch.

Your suggestion to turn off skinning worked correctly, thank you!

You can close this if you think me expecting the original model to export was incorrect. Or you can leave it open if you want to make "badly skinned" models still export correctly.

Is there a need for some kind of "virtual Rob" feature where you say, "tell me what Rob would say about this model", and OpenVSP would say, "skinning with strength 0 is not a good plan, maybe you should turn off skinning?"

ramcdona, 2023-09-25:
@jeffallen Sorry, the Fixes branch I was referring to was on my personal repo -- not the 'official' OpenVSP one....

https://github.com/ramcdona/OpenVSP/tree/Fixes

Does your version of OpenVSP not have a 'Virtual Rob' feature?  Mine does...  It seems to be able to communicate telepathically with my subconscious.  Shame it isn't working for you.  I'll look into it.

@cibinjoseph I posted on LI a while ago about a feature related to your suggestion...  https://www.linkedin.com/posts/rob-mcdonald-aerospace_aiaaaviation-activity-7073906623655677952-khdC/

jeffallen, 2023-09-26:
The fixes on your Fixes branch did not solve this.

ramcdona, 2023-09-26:
OK, thanks for the update.  The fix is a good one and might help with how your file is understood by downstream tools.

closed 2023-09-26 by ramcdona