OpenVSP/OpenVSP #279: Can't install on Debian 12

fsneeringer, 2023-10-04:
Running Debian 12 stable.  I run into a dependency error for python 3.10, however python 3.11 is installed.  See below:

```
:~/Downloads$ sudo dpkg -i OpenVSP-3.35.3-Ubuntu-22.04_amd64.deb 
Selecting previously unselected package openvsp.
(Reading database ... 244741 files and directories currently installed.)
Preparing to unpack OpenVSP-3.35.3-Ubuntu-22.04_amd64.deb ...
Unpacking openvsp (3.35.3) ...
dpkg: dependency problems prevent configuration of openvsp:
 openvsp depends on libpython3.10 (>= 3.10.0); however:
  Package libpython3.10 is not installed.

dpkg: error processing package openvsp (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openvsp
```

I then run: 
```
sudo apt install python3-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.11.2-1+b1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 openvsp : Depends: libpython3.10 (>= 3.10.0) but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

This shows a newer version (>= 3.10.0) installed but not recognized.  

Is there some way to allow install with newer than called out dependency?  I figured this would be allowed due to the dependency message I'm getting.

I tried compiling and installing by following directions posted but have gotten errors with that method as well, I can post if the above isn't the right path for installing on Debian 12.  Please let me know if I should post any more info.

Thanks!

cibinjoseph, 2023-10-04:
It's interesting that Ubuntu 22 comes with Python 3.10 and Debian 12 comes
with Python 3.11. You could try forcing python 3.10 to be installed on your
system following this thread
<https://askubuntu.com/questions/1402456/unmet-dependencies-for-python-3-10>
.
If you can't install python 3.10 libraries, you wouldn't be able to install
it using the precompiled deb package for Ubuntu 22.

What errors do you get when you try compiling it using the wiki
installation instructions? I would suggest posting those errors and seeing
if we can debug that installation route.

Cibin

fsneeringer, 2023-10-05:
Hi Cibin,

Thanks for the response.  From using the directions here: [Install Instructions](https://openvsp.org/wiki/doku.php?id=ubuntu_instructions) 

From step 4 make -j8 finishing lines are: 
```
-- Set runtime path of "/home/frank/OpenVSP/buildlibs/STEPCODE-prefix/src/STEPCODE/../sc-install/bin/lazy_sdai_ap203" to "/usr/local/lib:$ORIGIN/../lib"
-- Installing: /home/frank/OpenVSP/buildlibs/STEPCODE-prefix/src/STEPCODE/../sc-install/bin/lazy_sdai_ap203-static
-- Set runtime path of "/home/frank/OpenVSP/buildlibs/STEPCODE-prefix/src/STEPCODE/../sc-install/bin/lazy_sdai_ap203-static" to "/usr/local/lib:$ORIGIN/../lib"
-- Installing: /home/frank/OpenVSP/buildlibs/STEPCODE-prefix/src/STEPCODE/../sc-install/share/man/man1/fedex.1
-- Installing: /home/frank/OpenVSP/buildlibs/STEPCODE-prefix/src/STEPCODE/../sc-install/share/man/man1/exp2cxx.1
[ 97%] Completed 'STEPCODE'
[ 97%] Built target STEPCODE
make: *** [Makefile:91: all] Error 2

```

From cmake in step 5:
```
buildlibs -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.4") found components: doxygen dot 
-- CMINPACK_LIBRARY /usr/lib/x86_64-linux-gnu/libcminpack.so
-- Found CMINPACK: /usr/include/cminpack-1  
-- Found Code-Eli version 0.3.6, and requested version 0.3.6
-- Found CodeEli: /home/frank/OpenVSP/buildlibs/CODEELI-prefix/src/CODEELI-build/include;/home/frank/OpenVSP/buildlibs/CODEELI-prefix/src/CODEELI/include (Required is at least version "0.3.6") 
-- Found CPPTest: /home/frank/OpenVSP/buildlibs/CPPTEST-prefix/include  
CMake Error at cmake/Finddelabella.cmake:10 (FIND_PATH):
  Could not find DELABELLA_INCLUDE_DIR using the following files: delabella.h
Call Stack (most recent call first):
  external/CMakeLists.txt:18 (FIND_PACKAGE)

```
Please let me know if there's other info or logs that would be helpful for me to post.  I'm very much a beginner at building programs so I'm not familiar with good troubleshooting methods.

Thanks,
Frank

cibinjoseph, 2023-10-06:
Sorry, I couldn't set up a virtual machine to see what was going wrong.
I'll hopefully get to it this weekend.
Meanwhile, I suggest posting this in the OpenVSP google forum
<https://groups.google.com/g/openvsp>. Rob might know a quickfix without
much testing.

app4soft, 2023-10-08:
> Meanwhile, I suggest posting this in the OpenVSP google forum <https://groups.google.com/g/openvsp>. Rob might know a quickfix without much testing.

Here is a thread and it seems like not fixed yet:

* https://groups.google.com/g/openvsp/c/L6l8zibCuRE

Would be good if there would be a DEB-package for _Debian 12_ by OpenVSP team.

fsneeringer, 2023-10-09:
> Here is a thread and it seems like not fixed yet:
> 
> * https://groups.google.com/g/openvsp/c/L6l8zibCuRE

That is the thread I opened on Cibin's suggestion.  Rob was very helpful and suggested a fix to this and stated that longer term will incorporate the change so this won't be a problem again.  I was able to make the changes he suggested and build successfully.  Thanks everyone for your help.

Frank

app4soft, 2023-10-09:
> I was able to make the changes he suggested and build successfully. Thanks everyone for your help.
> 
> Frank

@fsneeringer, May I ask you to provide a DEB for Debian 12? Or at least command(s) you used for build & install OpenVSP under Debian 12?

fsneeringer, 2023-10-10:
> @fsneeringer, May I ask you to provide a DEB for Debian 12? Or at least command(s) you used for build & install OpenVSP under Debian 12?

I cannot provide a DEB for this as I didn't make one.  I followed Rob's directions in the other thread here:https://groups.google.com/g/openvsp/c/L6l8zibCuRE . To do this do the following.

Follow the build instructions linked here: https://openvsp.org/wiki/doku.php?id=ubuntu_instructions

On step 4 the build process will fail.  At this point, navigate to the directory you're building in and edit `buildlibs/DELABELLA-prefix/src/DELABELLA/delabella.cpp` and add `#include <time.h>` at the top of the file (without the quotes) using your preferred text editor.  Make sure to keep the <> around time.h. 

Go back to the directory in step 4 and re-run make.  

Continue with the remainder of the build instructions on the linked page. 

This worked for me and I can run the executables for OpenVSP. 

Rob did a much more thorough job of explaining in the google groups thread linked earlier.  Please reference that if you're looking for more information.  Hope this is helpful.

closed 2023-10-17 by fsneeringer