OpenVSP/OpenVSP #283: VSP Hangar

jphkun, 2023-12-15:
Hi everyone,

Not sure that it's the right place to post this, but I am trying relentlessly to access the OpenVSP Hangar.
Sadly the website is down since quite some time now.

Would it be possible add an "Example" folder at the root or lower, in the source code and ship the examples with the program?
People could add their ".vsp3" files on github while the website is down.

Other possibility, create a new repository with all the example files and people could just download this.
Just an idea.

I don't know how much you guys have on your plate at the moment, just an suggestion from an enthusiast :)

Kind regards,
Jean-Philippe

cibinjoseph, 2023-12-15:
I don't think bundling user-submitted vsp3 files with the OpenVSP package is a good thing. At this point, it's better to just make a new website.
You'll find more responses on the OpenVSP Google group though.

chiemcke, 2024-02-06:
Looks like the Hangar is still down (will not open in Chrome or Edge), and I have tried multiple times over the past two weeks.  Thanks, Christoph

trilbytim, 2024-03-07:
This comes up on the Google Groups very frequently. VSP Hangar is down and is probably not coming back (search the Google Groups for more information about this). 

It has been proposed to instead use a Git repository to store user generated models. This may in the future become officially supported but in the meantime there is an unofficial repository here: https://github.com/danielenriquez59/vsp-gmr/tree/master

ramcdona, 2024-11-10:
Hangar has been replaced by the 'airshow'.  

https://airshow.openvsp.org/

closed 2024-11-10 by ramcdona