OpenVSP/OpenVSP #288: Panel mode analysis not working through Python API

trilbytim, 2024-02-08:
When calling a Panel mode analysis through the Python API the analysis does not run due to a failure to find critical files, giving an error message such as:
`ERROR 7: Could not open History file: /home/timbo/Documents/Avian/CFD/elliptical test/VSP/planktest/planktest_DegenGeom.history
	File: /home/runner/work/OpenVSP/OpenVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2236
ERROR 7: Could not open Polar file: /home/timbo/Documents/Avian/CFD/elliptical test/VSP/planktest/planktest_DegenGeom.polar
	File: /home/runner/work/OpenVSP/OpenVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2486
ERROR 7: Could not open Load file: /home/timbo/Documents/Avian/CFD/elliptical test/VSP/planktest/planktest_DegenGeom.lod
	File: /home/runner/work/OpenVSP/OpenVSP/src/geom_core/VSPAEROMgr.cpp 	Line:2729

Error: Aerothermal database (*.adb) file not found. Execute VSPAERO before running the CpSlicer`

**Steps to reproduce:**
Demonstration script and test file is here:
[planktest.zip](https://github.com/OpenVSP/OpenVSP/files/14215056/planktest.zip)
When used through the Open VSP GUI, both VLM and Panel modes successfully solve for this example. However when running through the Python script it will exit with the above error messages. If the commenting is changed on lines 18 and 19 to change the analysis mode to VLM then it does solve correctly.
NOTE: If a VLM analysis is performed first and then the Panel mode is run, then it does run without errors. However on examining the output it can be seen that it's in fact just performing another VLM run.

System: Ubuntu 22.04, Python 3.11 running in Anaconda environment
OpenVSP version: 3.36 (this appears to have been broken somewhere between v3.31 and v3.36, because it was previously working)

christospliakos, 2024-11-11:
Have you found any solution on that?

HB-Stratos, 2025-01-11:
I'm also currently running into this problem. Have there been any news?

ketakopter, 2025-06-24:
This is a problem of how you set up the analysis: if you want to run Panel mode, you have to set it _twice_: one for VSPAEROComputeGeometry, and again for VSPAEROSweep. If you add `vsp.SetIntAnalysisInput(analysis_name, "AnalysisMethod", [vsp.PANEL])` after line 24, the script runs correctly. Tested with OpenVSP 3.43.

Basically, in your script you are telling to run the ComputeGeometry in PANEL mode, but then set the inputs of Sweep from whatever values are specified in the vsp3 file, which in this case is VLM. So you run ComputeGeometry in Panel mode, and then Sweep in vortex lattice, which fails.

closed 2025-07-19 by ramcdona