OpenVSP/OpenVSP #293: CFDmesh fails to complete with a simple element creation

mike15083, 2024-04-30:
Creating a mesh with a finite thickness trailing edge fails to complete what appears to be a simple element creation. 

[image]

This wing fails in the corner as shown below:
[image]

and at the symmetry plane
[image]

Process to demonstrate the issue:
Open the attached file
select CFDmesh
select mesh and export
[wingmesh bug 2.zip](https://github.com/OpenVSP/OpenVSP/files/15169464/wingmesh.bug.2.zip)

ramcdona, 2024-05-11:
This file does not correspond to the images shown above.  When I open this file, I only get a stubby center section to a wing, not a reasonable aspect ratio wing shown in the top image.

The attached file does have some holes in the mesh, but the meshing resolution parameters are pretty absurd.  When I change them to something reasonable, I reliably get watertight meshes.

For example, the growth ratio in the attached file is 1.75 -- The default is 1.3.  Most CFD experts consider that to be very coarse and instead prefer to have a growth ratio of about 1.15.

The maximum element size is 1.75 and the minimum is 0.5 -- the perspective, the span and chord are about equal and are about 4.0.  So, the finest the mesh should be allowed is about eight segments across the span or the chord.

Meanwhile, this model has a blunt trailing edge with a thick ness of 0.021.  This will get split into two, so there will be edges there that are only .0105 long.

If you have a particular case that you want me to look at, please be sure the file matches the screenshots and matches the bug report.

mike15083, 2024-06-05:
Rob,

I tried to create a smaller model for you and something went differently, sorry.
Here is a more representative model of what I am trying to get a reasonable mesh. I am not using RANS or Panel methods as the solver, so mesh guidelines are very different.
[image]
The mesh fails to complete at the wingtip on both sides as shown here:

[image]
I am trying to create a process that generates a mesh without user intervention, given an arbitrary set of mesh parameters.

mike15083, 2024-06-05:
[Mesh failure.zip](https://github.com/user-attachments/files/15597435/Mesh.failure.zip)