OpenVSP/OpenVSP #296: Ubuntu 24.04 packages?

lkarlslund, 2024-05-14:
Since Ubuntu 24.04 LTS is released, is there any chance of precompiled packages support for that platform?

I tried compiling it myself, but failed miserably with endless problems - probably no biggies but just my lack of experience with the build chain.

ramcdona, 2024-05-14:
I saw that GitHub Actions claims to have 24.04 images and tried to set up a corresponding build.  However, when launched, it waited forever and never got a runner to execute on.  So right now, I don't think it is really an available platform on the build robot.  When that changes, we can reconsider.

Unfortunately, I don't know when to expect that -- or how to know it has happened.

The automated build script should give you a good step-by-step for how to build

https://github.com/OpenVSP/OpenVSP/blob/main/.github/workflows/build.yml

In the meantime, there is a high probability that the 22.04 build will work fine on 24.04.

Rob

lkarlslund, 2024-05-15:
Thanks for the reply.

The 22.04 built .deb does not install on the 24.04 version, as it requires libpython3.10 which is unavailable (it has 3.11 and 3.12 on the current distro packages).

I tried forking OpenVSP to https://github.com/lkarlslund/OpenVSP/ and created a manually triggerable action to build using the Ubuntu 24.04 runners (I got one immediately).

Several issues:

"pip install sphinx" gives error what you can't do that in a distribution controlled python environment. I switched to "apt install python3-sphinx" to get that installed.

Later on I get the same error I end up with on my local system:

`/home/runner/work/OpenVSP/OpenVSP/buildlibs/EXPRPARSE-prefix/src/EXPRPARSE/src/exprparse/exprparse.cpp:62:9: error: ‘uint16_t’ does not name a type
   62 |         uint16_t precedance;
      |         ^~~~~~~~`

My skill-set ends there :-)

The action file is at https://github.com/lkarlslund/OpenVSP/blob/main/.github/workflows/build2404.yml

ramcdona, 2024-07-20:
Ubuntu 24.04 packages now created with version 3.40.0

closed 2024-07-20 by ramcdona