OpenVSP/OpenVSP #299: crash in 3.40.0: vector index out of bounds

fernape, 2024-07-23:
Platform: FreeBSD 14.1
OpenVSP version: 3.40.0

I am the maintainer of OpenVSP in FreeBSD and I am trying to update to 3.40.0. The build is fine and I can boot the application. However, when trying to open a file, OpenVSP crashes with this message in the console:
```
error : Unknown IO error
[LWP 101888 of process 7804 exited]

Thread 1 received signal SIGILL, Illegal instruction.
```

If I build with debugging symbols, the problem seems to be in the `void PropGeom::UpdateBladeAzimuth()` [function](
https://github.com/OpenVSP/OpenVSP/blob/b0f7b5fd4583a5c721c5087642cecb37e8f59842/src/geom_core/PropGeom.cpp#L1427): the initial loop it is trying to access `m_BladeAzimuthParmVec` beyond the end of the vector:

```
(gdb) p m_BladeAzimuthParmVec
$2 = std::vector of length 1 = {0x806df4fc0}
(gdb) p m_BladeAzimuthParmVec.size()
$3 = 1
```

```
_LIBCPP_ASSERT_VALID_ELEMENT_ACCESS(__n < size(), "vector[] index out of bounds")
```

This seems to happen with files like `5868_9_GEOM.vsp3` and `SADREY.vsp3` downloaded from the hangar.

ramcdona, 2024-07-25:
Thanks much!

closed 2024-07-25 by ramcdona

fernape, 2024-07-26:
Hi @ramcdona,

The crash is fixed but now, when trying to open any of the two mentioned files, vsp just uses 100% of 1 CPU core.

Do you want me to open a new issue for this?

ramcdona, 2024-07-26:
Are you checking a debug build, or a normal release build?

OpenVSP is extremely slow on a true release build.  It is not recommended.

How long does it use 100% of one cpu?  Why is this a problem?

fernape, 2024-07-27:
> Are you checking a debug build, or a normal release build?
> 
> OpenVSP is extremely slow on a true release build. It is not recommended.

Yes, that was it. Thanks!

> 
> How long does it use 100% of one cpu? Why is this a problem?

ramcdona, 2024-07-27:
Debug builds disable all compiler optimizations.  OpenVSP uses a lot of templated code, which is spectacularly bad if it is not optimized.  Consequently, it is totally unusable in Debug mode.

To get around this, CMake supports a ReleaseWithDebugInfo build -- that still has optimizations, but includes debug information.  This runs at nearly full speed and still lets you use debug and other instrumentation.  I recommend you use this if you're debugging something.

Since it still does optimizations, it will optimize out some variables and will change the order of execution in certain cases -- so the debug stepping might not be 100% perfect, but it will at least be usable.

Another good tool is to build with the address sanitizer.  It works really well at finding problems.