OpenVSP/OpenVSP #311: 'vsp.SetParmVal(wid, "Sym_Rot_N", "XForm", int(3) )' doesn't work on the model

RebelYoung, 2024-11-28:
I use the python api for automated generate models.
However, it doesn't work for the model by `'vsp.SetParmVal(wid, "Sym_Rot_N", "XForm", int(3) )'`

I have tried other variable on `'XForm'` like `'Z_Rel_Location'`, `'X_Rel_Rotation'`,etc.
All of them work efficiently.

There is no error when execute 'vsp.SetParmVal(wid, "Sym_Rot_N", "XForm", int(3) )', but `vsp.GetParmVal(wid, "Sym_Rot_N", "XForm")` return `0.0` not `3`, which original value is `11` in the based `.vsp` file.

[image]

[image]

RebelYoung, 2024-11-28:
I tried `vsp.SetParmVal(vsp.FindParm(wid,'Sym_Rot_N','XForm'),int(3))` , but it also didn't work.
`vsp.GetParmVal(wid,'Sym_Rot_N','XForm')` returned `0.0` not `3`.
[image]

RebelYoung, 2024-11-29:
I found `vsp.FindParm(wid,'Sym_Rot_N','XForm')` return `''`, not Parm ID.
So is there no `'Sym_Rot_N'` value in openvsp python script ?
[image]

ramcdona, 2024-11-30:
Questions like this are generally best asked in the OpenVSP Google Group.

Try:

`vsp.SetParmVal(wid, "Sym_Rot_N", "XForm", int(3) )
`

Note that the parameter group is `Sym` not `XForm`.  You can see this in the second image where you pulled up the information about the `Sym_Rot_N` parm.

closed 2024-11-30 by ramcdona