OpenVSP/OpenVSP #315: Bug: CFD Mesh crashing OpenVSP

vasylyamnych, 2025-01-07:
OS: Arch Linux, KDE, Nvidia
Reproduciable on 2 PCs.

OpenVSP 3.41.1
https://aur.archlinux.org/packages/openvsp

Steps
Analysis -> CFD Mesh... 

/usr/include/c++/14.2.1/bits/stl_vector.h:1130: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = DrawObj; _Alloc = std::allocator<DrawObj>; reference = DrawObj&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    8358 IOT instruction (core dumped)  vsp

ramcdona, 2025-01-10:
Is this reproducible on an empty file, or with a particular model loaded?  If it is with a particular model loaded, can you provide that model?

vasylyamnych, 2025-01-10:
it is reproducible on both - empty file, a basic wing...

mike15083, 2025-02-17:
I have a similar problem with a clean wing in 3.42.2

[Short wing test.zip](https://github.com/user-attachments/files/18828280/Short.wing.test.zip)

ramcdona, 2026-02-03:
I do not have access to an Arch Linux machine (or VM) to test on.

I am not seeing this problem on the Linux builds I do have access to.

Someone needs to do some more testing to help narrow this down.  My gut would say it could be a problem with the gcc version used to build on Arch vs. other Linux platforms.

Someone should build a RelWithDebugInfo version and see if the crash stack trace provides any more information.

ramcdona, 2026-10-10:
There has been a lot of work on CFDMesh.  Please try this on 3.54.0 or newer and reopen if there are still problems.

closed 2026-10-10 by ramcdona