OpenVSP/OpenVSP #316: Crash when launching GUI from python API: Current driver does not support Vertex Buffer Object

HB-Stratos, 2025-01-10:
I've been learning about how to use the python API, and it seems I've finally figured out how to create and modify geometry even without much documentation (unless I missed where that is located for python, had to rely on google groups threads for examples).

Either way, I'm now trying to run the GUI to visualize my changes easily. However, the GUI crashes as soon as it loads more than a white empty window with the following output:

```
Help files found
Current driver does not support Vertex Buffer Object.
OpenGL version reports: 4.6.0 NVIDIA 560.94
Current driver does not support Vertex Buffer Object.
OpenGL version reports: 4.6.0 NVIDIA 560.94
Current driver does not support Vertex Buffer Object.
OpenGL version reports: 4.6.0 NVIDIA 560.94
Current driver does not support Vertex Buffer Object.
OpenGL version reports: 4.6.0 NVIDIA 560.94

Process finished with exit code -1073741819 (0xC0000005)
```

here's the code that is generating this crash:

```py
import os.path
from time import sleep

import openvsp_config
openvsp_config.LOAD_GRAPHICS = True # if enabled load the GUI
openvsp_config.LOAD_FACADE = False # if enabled load communication layer and set OpenVSP to separate python process
import openvsp as vsp

# Very helpful example code:
# https://groups.google.com/g/openvsp/c/GLRyu0vVEDY

vsp.VSPRenew() # clear everything
vsp.VSPCheckSetup()

wing_id = vsp.AddGeom("WING")
vsp.SetParmVal(wing_id, "TotalSpan", "WingGeom", 1.4)

vsp_file_path = os.path.join(os.getcwd(), __file__+".vsp3")
vsp.WriteVSPFile(vsp_file_path)

vsp.InitGUI()
vsp.StartGUI() # <--- this is where it crashes. The window pops up for a second all white, then disappears with the errors above

#vsp.PopupMsg("This is a test") # <--- if uncommented this also throws an error, presumably because the GUI crashed.
```
Setting `LOAD_FACADE`to True still crashes, but the process exit code becomes zero, presumably because the crash occurs in another thread.

Am I doing something wrong here, or did I stumble across a bug?

HB-Stratos, 2025-01-11:
I have figured something out: Moving the `vsp.InitGUI()` and `vsp.StartGUI()` to right under `import openvsp as vsp` works, then the UI starts up just fine. Only `InitGUI()` up there and `StartGUI()` somewhere further down does not work, both must be right at the top of the script.

ramcdona, 2025-01-11:
You've correctly found the workaround.  This bug has been fixed and will go out in the next release.

closed 2025-01-11 by ramcdona