OpenVSP/OpenVSP #323: Aero Viewer doesn't work in 3.42.3

jknight-fhe, 2025-03-06:
Just updated to OpenVSP 3.42.3

After I run a small VSPAERO run then hit "Launch Viewer", nothing happens and I get the message in the viewer console:
"C:\Program Files\OpenVSP-3.42.3-win64\vspviewer.exe" "< my local path>"
Could not open the font file for writing! 

Running the exact same file and analysis in OpenVSP 3.40.1, the viewer works just fine.  

Looks like this was a problem in 2015, having to do with the model file being in a separate directory from main OpenVSP directory.

dmstan, 2025-03-07:
This seems to be related to write permissions, see this: https://github.com/OpenVSP/OpenVSP/issues/305

ramcdona, 2025-07-19:
This should be fixed in 3.45.0.  Please re-open if problems persist.

closed 2025-07-19 by ramcdona