OpenVSP/OpenVSP #326: Ubuntu 25.10 Issue

bigg-g-makes, 2025-04-26:
I am running Ubuntu 25.10 and I am experiencing an unmet dependency issue, I did try sudo apt --fix-broken install but did not solve this issue.  Could this be a python issue?  Any assistance would be apreciated.

[image]

bigg-g-makes, 2025-04-26:
I have also tried to compile from source and still no joy

ramcdona, 2025-05-02:
Did you try installing the libcminpack, libglew, and libpython packages?

bigg-g-makes, 2025-05-02:
> Did you try installing the libcminpack, libglew, and libpython packages?

It is a depencency nightmare, fix one, then another, and another.... so far its one by one by one but no joy still

closed 2025-07-19 by ramcdona