OpenVSP/OpenVSP #329: OpenVSP under Debian 12.10, gcc (GCC) 14.3.0 : compilation error /usr/include/c++/12/bits/std_thread.h:144:13: error: cannot convert ‘long int’ to ‘void (*)()’

jaque-carambole, 2025-06-06:
Hi, I tried to compile OpenVSP under Debian 12.10, gcc (GCC) 14.3.0 following the instructions of
https://openvsp.org/wiki/doku.php?id=ubuntu_instructions&s[]=debian
compilation fails at step 6 Build OpenVSP
with some errors  (2)

__/usr/include/c++/12/bits/std_thread.h:144:13: error: cannot convert ‘long int’ to ‘void (*)()’
  144 |             __depend);
      |             ^~~~~~~~
      |             |
      |             long int__

full trace of installation run attached

[traceAll.txt](https://github.com/user-attachments/files/20630671/traceAll.txt)

Thanks for your help!
Best regards

baranidesign, 2025-06-09:
Thank you for your email. Due to current commitments, I check messages
only once or twice a week. For a faster response, please reach out to
BARANI DESIGN Technologies via our official channels. I appreciate
your patience and will get back to you as soon as I’m able.

ramcdona, 2025-07-19:
Please reopen and update if you're still having problems like this.  Be sure to specify which version of OpenVSP you are trying to build.

closed 2025-07-19 by ramcdona