OpenVSP/OpenVSP #338: Can't run vsp.exe

chaouil77, 2025-09-20:
New to openVSP,
Was trying to install it and whenver i click on vsp.exe to run it, I get a "ceased to function" error. I cant even troubleshoot a specific error, so I apologize...i really need the help

ramcdona, 2025-09-20:
What version of OpenVSP are you trying to run?  Have other prior versions worked for you on this computer?

If this is the first time you've tried OpenVSP on this computer, it is most likely a problem that you do not have an appropriate graphics card driver installed for your computer.  OpenVSP requires OpenGL for 3D graphics.  Recent versions of Windows no longer support OpenGL with the Microsoft supplied drivers.  Get a graphics driver from your chip manufacturer and that should help.

baranidesign, 2025-09-21:
Thank you for your email. Due to current commitments, I check messages
only once or twice a week. For a faster response, please reach out to
BARANI DESIGN Technologies via our official channels. I appreciate
your patience and will get back to you as soon as I’m able.

ramcdona, 2025-09-21:
Go to AMD's web site, go to the Support area, go to Downloads.  Put in the information for your graphics card and operating system.  It will provide a file to download.  Download that file, unzip or run it, run the installation program contained inside.  Follow the instructions to install the graphics driver.

chaouil77, 2025-09-21:
I install the the driver for my graphic card and still meet the exact same issues unfortunately